wpilibsuite/ntcore #12: Builds fail with Visual Studio 2015.

ThadHouse, 2015-08-22:
On my system I only have VS 2015 installed, and I get a linker error when gradle tries to compile the program.

```
See file:///C:/Users/thad/Documents/GitHub/ntcore/build/tmp/linkX64NtcoreSharedLibrary/output.txt for all output for linkX64NtcoreSharedLibrary.
linking ntcore.dll failed.
LINK : fatal error LNK1104: cannot open file 'libcpmt.lib'

Finished linkX64NtcoreSharedLibrary, see full log file:///C:/Users/thad/Documents/GitHub/ntcore/build/tmp/linkX64NtcoreSharedLibrary/output.txt.
```

I noticed that looking in the options.txt file, it puts the VS directory to `"/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\lib\amd64"`, which is for VS2013. Is it supposed to be detecting VS 2015 and switching to it?

PeterJohnson, 2015-08-23:
@333fred, this is the VS 2015 gradle build issue.

333fred, 2015-08-23:
Gradle currently has issues with VS2015, so this is expected. I actually detect if you have VS 2015 and switch to 2013, I should probably add checking if you only have vs2015 and erroring out.

PeterJohnson, 2015-09-06:
Check added in 21e21d3b8bf6f99de236d2eb41d895403f75d7ff.

closed 2015-09-06 by PeterJohnson