wpilibsuite/ntcore #13: No entry points found for Windows DLL's.

ThadHouse, 2015-08-22:
I was able to get a successful build on a VS 2013 machine, but when trying to open the DLL it says it cannot find the entry point its looking for. Running dumpbin on the compiled DLL shows no functions at all being exported from the DLL. Are there any special gradle flags needed to get the exports to build correctly?

PeterJohnson, 2015-08-22:
I've always used cmake to build on Windows.  The ntcore.def looks like it's being used in gradle too, but maybe something's broken there.  @333fred, help?

ThadHouse, 2015-08-22:
The CMake build does work. I did find a typo in the ntcore.def file for the NT_PostRpcResponse function, which was NT_PostRpcRepsonse. I added that fix to the pull request, which fixed my cmake builds, but it did not fix the gradle build.

333fred, 2015-08-22:
Sorry for the lack of responses guys. I'm currently transferring planes,
I'll be able to look tomorrow. As to msvc 2015, gradle has some issues with
it and doesn't include the right headers, causing compilation issues.

The CMake build does work. I did find a typo in the ntcore.def file for the
NT_PostRpcResponse function, which was NT_PostRpcRepsonse. I added that fix
to the pull request, which fixed my cmake builds, but it did not fix the
gradle build.

333fred, 2015-08-23:
Thad, I've just pushed a pull request, #17, which should fix the gradle build.

commit 9d4a66a4 (2015-08-23): Merge pull request #17 from 333fred/master

Fixed Gradle build to actually export proper functions

Fixes #13.
closed 2015-08-23 by PeterJohnson