wpilibsuite/ntcore #14: Setting a value does not notify listeners on the setting system.

ThadHouse, 2015-08-22:
In previous implementations, if you set a value, it would notify all the listeners on the setting system. This implementation does not seem to be doing that, and only notifying either when immediate_notify is true, or when processing an incoming message.

PeterJohnson, 2015-08-22:
This was an intentional change.  The issue with the old behaviour is it's easy to unintentionally cause a update loop and/or erasure of local changes.  For example, take a look at PIDController.  SetPID() updates the PID values and also updates the table.  It's also listening for changes to the table in ValueChanged(), and if they're different from the current values, SetPID() is called again.  If you call SetPID() multiple times, you could end up losing changes (because the second call could get reset when the notifier finally fires on the first call), and/or have strange multiple updates occurring.

ThadHouse, 2015-08-22:
Ok. Closing.

closed 2015-08-22 by ThadHouse