wpilibsuite/ntcore #15: Out of memory checks

PeterJohnson, 2015-08-23:
Check for null returns in the few places that use malloc to handle out of memory errors (relatively) gracefully.

XavilPergis, 2018-03-13:
So I've been looking over this and adding in a few checks
- Functions that return arrays just need a check to see if `malloc` failed, returning `nullptr` if it did
- Functions that return 2D arrays need the same check for the top-level array, as well as checks for each sub-arrays. If an allocation of a sub-array fails, either all the entries before the current one can be freed and `nullptr` returned, or that entry set to `nullptr`. Freeing everything and returning `nullptr` seems to assume less about the caller, though, and I think this is the safest behavior.

Things get a little bit hairy with the `ConvertToC` functions with checks, because then a struct could be partially initialized without knowing so. Example:
```cpp
static void ConvertToC(const EntryInfo& in, NT_EntryInfo* out) {
  // Returning early here possibly leaves `out` completely uninitialized,
  // but the caller doesn't know
  if (!out) return false;
  out->entry = in.entry;
  // out->name could be uninitialized here, but we don't know
  // and can't propagate that error up to the caller
  ConvertToC(in.name, &out->name);
  out->type = in.type;
  out->flags = in.flags;
  out->last_change = in.last_change;
}
```
Since `ConvertToC` is a private interface, it *is* possible to propagate error information, but then you have the same problem when you get to the public interfaces; `NT_GetEntryValue`, for example, has no way to tell the caller that there was an allocation error other than leaving the value marked as unassigned. Doing that, though, would cause ambiguity between an OOM error and a nonexistent entry.

At the very least, the undefined behavior should be eliminated, and this could be done by just aborting the process if an alloc failed.

PeterJohnson, 2018-03-14:
Frankly, out of memory errors are seldom recoverable.  Simply aborting on allocation failure is probably the right answer.

XavilPergis, 2018-03-14:
Yeah, that seems like the most consistent thing to do. I'll submit a PR in a bit.