wpilibsuite/ntcore #16: NT_GetEntryInfo throws an Access Violation Exception

ThadHouse, 2015-08-23:
If there is at least 1 valid entry, the code throws an Access Violation when trying to call NT_DisposeString while converting the StringRef to an NT_String. The code mallocs for the NT_EntryInfo array, but the NT_String contained in that seems to point to garbage, which fails why trying to free it before copying.

```
testntcore.exe!check_bytes(const unsigned char * const first, const unsigned char value, const unsigned int size) Line 194	
testntcore.exe!is_block_an_aligned_allocation(const void * const block) Line 251	
testntcore.exe!free_dbg_nolock(void * const block, const int block_use) Line 870	
testntcore.exe!_free_dbg(void * block, int block_use) Line 1011	
testntcore.exe!free(void * block) Line 16	C++
testntcore.exe!NT_DisposeString(NT_String * str) Line 406	
testntcore.exe!nt::ConvertToC(llvm::StringRef in, NT_String * out) Line 141
testntcore.exe!ConvertToC(const nt::EntryInfo & in, NT_EntryInfo * out) Line 26
testntcore.exe!NT_GetEntryInfo(const char * prefix, unsigned int prefix_len, int types, unsigned int * count) Line 160
```

PeterJohnson, 2015-08-23:
It's probably a bad idea to dispose it in ConvertToC(NT_String), as the conditions under which that is really necessary are minimal.  Plus, better a memory leak than trying to free garbage.

ThadHouse, 2015-08-23:
I agree. It looks like all the other times that function is used, NT_InitString is called before hand. So we can either call InitString inside the function instead of dispose, or call nothing and require the caller to have already initialized the string.

commit 29f73cb5 (2015-08-23): Don't dispose in ConvertToC for NT_String and NT_Value.

Fixes #16.
closed 2015-08-23 by PeterJohnson