wpilibsuite/ntcore #19: Creating a client causes an Assertion Exception on Win32

ThadHouse, 2015-08-25:
In TCPConector.cpp, on line 76, it tries to append '\0', which causes the StringRef to fail, because it is seeing '\0' as null. I tried modifying that line to try and fix it, and there were some cases where I would get it to not throw the error, but I could never get it to connect to a server. Client on a RoboRIO did work however, so its limited to the Win32 builds.

PeterJohnson, 2015-08-25:
Should be able to change that line from append() to push_back().  Or better, use c_str() instead of data() on line 78 and delete line 76.  I'll commit the latter tonight, but this way you can test it before I get to it.

ThadHouse, 2015-08-25:
It required me to cast the c_str to an LPSTR. That fixes the assertion error, however it still will not connect to a server. I've tried both an NTCore server, a .NET native server, and a java server, and it will not connect to either of them. This is using both with and without a \n on the end of the IPAddress string. I know that others will connect because an old SmartDashboard connects properly to all 3 servers. Running the client on linux does work properly, so it looks like its limited to Windows clients.

commit a3dbe9a8 (2015-08-25): Fix Windows client.

Windows returns WSAEWOULDBLOCK on a connect() attempt on a nonblocking socket.

Also wrap socket error handling so errors are correctly reported on Windows.

Fixes #19.
closed 2015-08-25 by PeterJohnson