wpilibsuite/ntcore #20: Maven Central Repository

JLLeitschuh, 2015-08-31:
Would you be so kind as to publish this to the Maven Central Repository so that we can use it in our project.
https://github.com/WPIRoboticsProjects/GRIP

333fred, 2015-08-31:
Jonathan, I don't know as though we will ever want to publish this to maven central. We aren't currently with the rest of wpilib. Is there any reason that you can't use a local publish for this?

PeterJohnson, 2015-09-01:
This project is quite general purpose (unlike WPILib, it is not specific to FIRST and has no RoboRIO or FIRST-specific dependencies), so I can see the argument for publishing it in central repositories such as Maven.  However, I think we should move the repository from my user account to either a GitHub organization or to usfirst.collab.net before we do that type of publish... we should talk to Brad about what makes sense.

JLLeitschuh, 2015-09-02:
Yes, we are using Travis-CI for our continuous integration. Since travis is running out on the cloud there is no way to do a local install.

PeterJohnson, 2015-09-23:
Maven publish info committed in 75358e6c45805e709ff594e1bc6cbb96cbd7205c.

JLLeitschuh, 2015-09-24:
I'm not seeing it on maven central.
http://search.maven.org/#search%7Cga%7C1%7Cwpi
Also, there is an organization that may be appropriate to hand this off to.
https://github.com/WPIRoboticsProjects

333fred, 2015-09-24:
You wouldn't. This publishes to the local maven repository, not to Maven central. Also, this is now effectively a mirror of the version on collabnet.

JLLeitschuh, 2015-09-25:
Is there a repository that I can point gradle at to download the artifact into our project?

333fred, 2015-09-25:
Not currently. We're still working on the multi-platform build. For now, what I would suggest is running the publishToMavenLocal task, that will publish it to your local maven repository. Still requires you to checkout the project, but better than nothing.

JLLeitschuh, 2015-09-25:
I can't build on my local machine (doesn't work on a mac) and I can't get Travis CI to do that either.

333fred, 2015-09-25:
Mac is a known issue. What's the problem with Travis? I can't help if I can't see a log.

JLLeitschuh, 2015-09-25:
I would have to setup Travis to download and  build the Ntcore project
every time we wanted to run our build. That would also require that we had
the jdk for Linux arm downloaded automatically for every build.
Travis generates fresh virtual machine for every build. Setting up up
things so ntcore would be persistent would take more time.

333fred, 2015-09-25:
We've incorporated the headers into the project, so downloading the jdk is no longer required.

333fred, 2015-12-20:
ntcore is now published to the FRC Maven repo. See the README for the artifact identifiers.

closed 2016-01-22 by PeterJohnson