wpilibsuite/ntcore #30: Server crashes upon exit if RPC callbacks exist.

ThadHouse, 2015-11-10:
If an RPC callback has been created, the library segfaults on exit. It seems like this is caused by the destructor for Storage.cpp notifying all the callbacks. However by that time, its very possible that the callbacks don't exist anymore. Is there a reason that the destructor for that is notifying the RPC's?

PeterJohnson, 2015-11-10:
The destructor doesn't notify the RPC callbacks, it notifies the condition variable that's used by GetRpcResult (which shouldn't even be called on the server).  It needs to do that otherwise you can end up with threads not terminating as they're hung up there.  However, I think there is a bug in GetRpcResult, in that it doesn't check m_terminating immediately after wait returns, but I don't know if that's related to the crash you're seeing.

ThadHouse, 2015-11-10:
I'm still trying to do more debugging. If I remove any tests that use entry listener callbacks or rpc callbacks, the program exits successfully. As soon as I add any tests that add callbacks to either of those, it segfaults on exit. When I get home tonight I'll try and write a c++ test for this, and see if I can reproduce it in entirely c++ code.

ThadHouse, 2015-11-10:
Also this is only reproducible currently on linux. However I believe that is caused by difference in Mono vs Microsoft .NET, and if I can reproduce it in C++ it should show up in both OS's.

ThadHouse, 2015-11-11:
I was able to grab a backtrace, and it seems to be a deadlock in Notifier.

```
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007fffef4e34bc in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007fffef84a531 in nt::Notifier::ThreadMain (this=0x7fffefabeba0 <nt::Notifier::GetInstance()::instance>)
    at /home/thad/git/peter/ntcore/src/Notifier.cpp:60
#3  0x00007fffef85144f in std::_Mem_fn<void (nt::Notifier::*)()>::operator()<, void>(nt::Notifier*) const (
    this=0xaaa5f8, __object=0x7fffefabeba0 <nt::Notifier::GetInstance()::instance>)
    at /usr/include/c++/4.8/functional:601
#4  0x00007fffef85139f in std::_Bind_simple<std::_Mem_fn<void (nt::Notifier::*)()> (nt::Notifier*)>::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=0xaaa5f0) at /usr/include/c++/4.8/functional:1732
#5  0x00007fffef8512a7 in std::_Bind_simple<std::_Mem_fn<void (nt::Notifier::*)()> (nt::Notifier*)>::operator()() (
    this=0xaaa5f0) at /usr/include/c++/4.8/functional:1720
#6  0x00007fffef851240 in std::thread::_Impl<std::_Bind_simple<std::_Mem_fn<void (nt::Notifier::*)()> (nt::Notifier*)> >::_M_run() (this=0xaaa5d8) at /usr/include/c++/4.8/thread:115
#7  0x00007fffef4e6a40 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff74b2182 in start_thread (arg=0x7fffee21c700) at pthread_create.c:312
#9  0x00007ffff71df47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

The Dispatcher and TCP acceptor threads seem to be properly stopping, however the notifier thread seems to be stuck.  This was grabbed after waiting for 15 or so seconds after the main method returned.

ThadHouse, 2015-11-11:
That was with an Entry Listener. Here is a backtrace when an RPC listener is enabled

```
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007fffef4e34bc in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007fffef842708 in nt::RpcServer::ThreadMain (this=0x7fffefabe9e0 <nt::RpcServer::GetInstance()::instance>)
    at /home/thad/git/peter/ntcore/src/RpcServer.cpp:111
#3  0x00007fffef847d25 in std::_Mem_fn<void (nt::RpcServer::*)()>::operator()<, void>(nt::RpcServer*) const (
    this=0xade3a8, __object=0x7fffefabe9e0 <nt::RpcServer::GetInstance()::instance>)
    at /usr/include/c++/4.8/functional:601
#4  0x00007fffef847c75 in std::_Bind_simple<std::_Mem_fn<void (nt::RpcServer::*)()> (nt::RpcServer*)>::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=0xade3a0) at /usr/include/c++/4.8/functional:1732
#5  0x00007fffef847b7d in std::_Bind_simple<std::_Mem_fn<void (nt::RpcServer::*)()> (nt::RpcServer*)>::operator()() (
    this=0xade3a0) at /usr/include/c++/4.8/functional:1720
#6  0x00007fffef847b16 in std::thread::_Impl<std::_Bind_simple<std::_Mem_fn<void (nt::RpcServer::*)()> (nt::RpcServer*)> >::_M_run() (this=0xade388) at /usr/include/c++/4.8/thread:115
#7  0x00007fffef4e6a40 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff74b2182 in start_thread (arg=0x7fffee21c700) at pthread_create.c:312
#9  0x00007ffff71df47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```
If you run a program with both entry listeners and rpc listeners, both threads hang.

PeterJohnson, 2015-11-11:
Do you have a test case program that you could share?

ThadHouse, 2015-11-11:
I'm trying to build one in C++ right now. Unfortunately it only happens on linux, and my c++ skills are awful. I should have one soon. However if it helps, the .NET code is below and the C++ with the same commands should be able to reproduce it. 

```
NetworkTable.SetIPAddress ("127.0.0.1");
            NetworkTable.SetPort (10000);
            NetworkTable.SetServerMode ();
            NetworkTable.Initialize ();

            System.Threading.Thread.Sleep (1000);

            NetworkTables.Native.CoreMethods.DeleteAllEntries ();

            CoreMethods.SetEntryString ("Key", "Value");

            NotifyFlags f = NotifyFlags.NotifyNew | NotifyFlags.NotifyUpdate;
            f |= NotifyFlags.NotifyImmediate;

            int listener = CoreMethods.AddEntryListener ("Key", (uid, key, value, flags_) => {
                    
            }, f);

            CoreMethods.SetEntryString ("Key", "Value2");

            Thread.Sleep (20);

            CoreMethods.RemoveEntryListener (listener);

            //Thread.Sleep (5000);

            Console.WriteLine ("Finished");

```

PeterJohnson, 2015-11-11:
The equivalent (https://gist.github.com/PeterJohnson/ebb35d04ead31b0ca20a) seems to work okay in C++, but debugging these kinds of things is hard.  I'll look a bit more at the backtraces you sent.

ThadHouse, 2015-11-11:
So I got somewhere. If I make it so it never actually runs the callback, it completes properly. So for some reason mono seems to be holding onto that semaphore. and not releasing it. I'm looking into seeing if thats an ntcore issue or a mono issue.

ThadHouse, 2015-11-11:
So turns out I was debugging this completely wrong. Mono waits for all foreground threads to exit before exiting the main thread. So the thread getting stuck is right. I still get this issue when running nunit, however I can't seem to reproduce it without nunit. So for now I'm going to close it. I don't think it's an ntcore related bug anyway.

closed 2015-11-11 by ThadHouse

ThadHouse, 2015-11-12:
Would you be opposed to adding StopRPC and StopNotifier methods to the native C interface? Or making StopServer and StopClient force those threads to exit as well. If I close those threads before the program exits, the segfault stops. I'll still try and debug if I can fix the issue the right way, but I'm guessing its either deep in the nunit or mono codebase, and thats going to take a LONG time to debug. Plus it seems like a good idea to stop those threads on shutdown anyway, or at least have a way to do so.

PeterJohnson, 2015-11-12:
No, I'd not be opposed to adding StopRpc and StopNotifier methods.  I think I want them to be separate from StopServer and StopClient, as those functions currently only affect network operations.

ThadHouse, 2015-11-12:
Ok. Ill write those tonight and submit a pull request.