wpilibsuite/ntcore #39: Race condition during dispatcher stop

ThadHouse, 2015-12-23:
I've been running into a weird race condition while trying to write unit tests, switching between client and server mode. If I start and stop the server side multiple times, then start and stop the client side, after a few client cycles the Stop() method seems to enter a race condition. When I break the code and look at the threads, it seems like a server thread is trying to grab the shutdown mutex, and the Stop() method is stuck in joining the clientserver thread, waiting for the thread to actually exit. It seems weird that a server is still running, but I wonder if this is happening because it's getting detached instead of joined in one of the earlier shutdowns. I'll put a monitor to detect if any of the early threads as I do this are being detached instead of joined, maybe that will help debugging. Also I've only noticed this on Windows, and if you actually set break points in the Stop() method the issue never occurs, which makes it even harder to debug.

PeterJohnson, 2015-12-23:
I have a fix about ready to push for this.  The issue is it's not properly getting accept shut down in the server thread.  I'm removing the detach case now that I have the join case robust, which should fix the issue you're seeing.

PeterJohnson, 2015-12-23:
Should be fixed by 44821c3e3c95bc6422284e5cfec2a4fef8494c63

closed 2015-12-23 by PeterJohnson