wpilibsuite/ntcore #42: ntcore client does not successfully connect to 2.0 servers

virtuald, 2016-01-03:
Using `OutlineViewer.jar` from the latest wpilib tools, I connected to a `OutlineViewer.jar` running a NetworkTables server, and received the following error messages from it on stdout.

```
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@3cdc0ee8 entered connection state: GOT_CONNECTION_FROM_CLIENT
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@3cdc0ee8 entered connection state: SERVER_ERROR: class edu.wpi.first.wpilibj.networktables2.connection.BadMessageException: Client Connected with bad protocol revision: 0x300
Close: edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@3cdc0ee8
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@2290f5d4 entered connection state: GOT_CONNECTION_FROM_CLIENT
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@2290f5d4 entered connection state: CONNECTED_TO_CLIENT
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@7e670420 entered connection state: GOT_CONNECTION_FROM_CLIENT
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@2290f5d4 entered connection state: CLIENT_DISCONNECTED
Close: edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@2290f5d4
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@7e670420 entered connection state: CONNECTED_TO_CLIENT
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@4a1cbe5e entered connection state: GOT_CONNECTION_FROM_CLIENT
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@7e670420 entered connection state: CLIENT_DISCONNECTED
Close: edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@7e670420
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@4a1cbe5e entered connection state: CONNECTED_TO_CLIENT
edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@4a1cbe5e entered connection state: CLIENT_DISCONNECTED
Close: edu.wpi.first.wpilibj.networktables2.server.ServerConnectionAdapter@4a1cbe5e
...
```

A similar error occurs when connecting to a Python robot using pynetworktables.

closed 2016-01-04 by PeterJohnson

virtuald, 2016-01-04:
Built OutlineViewer, it seems to work. Thanks!

PeterJohnson, 2016-01-04:
Note that while ntcore will connect to 2.0 servers, the new LabView dashboard will not, so it's still important to get 3.0 support into pynetworktables.  I'll work on that tomorrow.

commit 5e2a07d5 (2016-01-04): Fix client connection to 2.0 server.

Fixes #42.