wpilibsuite/ntcore #47: Supress Spammy output

JLLeitschuh, 2016-01-06:
```
NT: ERROR: select() to  port 1735 error 61 - Connection refused (ntcore-deskt:157)
NT: ERROR: select() to  port 1735 error 61 - Connection refused (ntcore-deskto:157)
NT: ERROR: select() to  port 1735 error 61 - Connection refused (ntcore-deskt:157)
NT: ERROR: select() to  port 1735 erroNT: ERROR: select() to  port 1735 error 61 - Connection refused (ntcore-deskto:157)
NT: ERROR: select() to  port 1735 error 61 - Connection refused (ntcore-deskto:157)
NT: ERROR: select() to  port 1735 error 61 - Connection refused (ntcore-deskto:157)
NT: ERROR: select() to  port 1735 error 61 - Connection refused (ntcore-deskto:157)
```
Is there any way to suppress this output?
Can we pipe it into an alternative stream?
This error is a normal part of the operation of GRIP and it cluttering our console log.

PeterJohnson, 2016-01-06:
You can install your own logger hook and direct the output however you want.  See https://github.com/PeterJohnson/ntcore/blob/master/java/test/Client.java for an example of doing this (the example sets the logging level to 0, which is not what you want; a level of 20 gives just INFO messages and above, 30 WARNING and above, etc).

333fred, 2016-01-06:
Talked with Jonathan offline. This is resolved.

closed 2016-01-06 by 333fred