wpilibsuite/ntcore #49: Gradle builds don't detect 32 bit only devices

ThadHouse, 2016-01-26:
I noticed when trying to perform a default gradle build on an arm device, which is 32 bit only, it would not compile correctly. The x64 portion of the build would sometimes complete and sometimes just hang for hours, and the x86 build automatically failed, because arm GCC does not have the -m32 flag. I don't have a 32 bit desktop system to test this on, but it would probably have the same issue. I don't know if gradle has an easy way to detect a 32 bit only system, and remove the -m32 flag. I was able to get an always compiling local arm build by running ```./gradlew :arm:build -PcompilerPrefix= ```, but thats not a valid hack for normal use.

333fred, 2016-01-26:
Check the toolchains/ folder. You can add/remove flags there, I think it should be pretty obvious how to add and remove flags. I don't have a 32-bit only system to test with myself, but the arm.gradle setup should be removing the -m32 flag already, which is why it works.

ThadHouse, 2016-01-26:
I'm not worried about building for me, as the :arm:build with the empty compiler prefix works just fine for me, was just worried about other people if they attempted to build it. There have been a lot of people trying to get NetworkTables working on offboard devices for camera tracking this year, and I know that other people are going to run into the same issue and not know how to work around it.

PeterJohnson, 2017-08-13:
Fixed by #210.

closed 2017-08-13 by PeterJohnson