wpilibsuite/ntcore #50: No support for transactions in the Java version of NetworkTable?

gartaud, 2016-02-05:
Hi
We are using NetworkTable to get information about potential targets from GRIP. The problem is that we need to retrieve multiple arrays of double as one transaction so that the information is coherent. It looks like NetworkTable implemented beginTransaction()/endTransaction() in the past but it appears it is no longer the case.
We believe this is why we sporadically get out-of-bound exceptions when sharing an index between the multiple arrays.
Thank you for looking into it.

333fred, 2016-02-05:
To my knowledge, NetworkTables has never implemented coherent methods like the ones you described. Specifically, I went back and checked both the Java and C++ implementations for something like begin/endTransaction, but I was not able to find anything. While that's certainly something we might be able to look into for future years, it's not something we will be able to add for this year.
Specifically, the idea of an atomic transaction in a networked protocol is very hard. Let's take the GRIP use case, for example. I'm assuming that you're referring to something like the contour report, which publishes multiple arrays of points. If you truly want a coherent update, not only does your retrieval of the values have to be coherent, but the actual publish of the values from client to server, and then to all other clients has to be coherent, or there is nothing stopping you from atomically retrieving both arrays when one has been updated and the other has not. There is currently nothing in the NetworkTables wire protocol to allow for this type of atomic transaction, so having a begin/endTransaction on either end would give the illusion of atomicity where none exists. I don't mean to sound dismissive, but with version 3.0 of the spec as it currently exists, you cannot effectively implement atomic transactions.
For your issue of getting IndexOutOfBoundsExceptions, the best I can tell you is to make sure that your arrays have the same length before blindly attempting to use the same index for both. Without protocol-level support for atomics, that's really your only option.

gartaud, 2016-02-05:
Thank you for your prompt reply.
It looks like it existed back in 2012
See http://wbrobotics.com/javadoc/edu/wpi/first/wpilibj/networktables/NetworkTable.html#beginTransaction()
I am indeed referring the a contour report that is sometimes incoherent, probably due to a race condition between us reading it and the report being updated as we do.
Looking into it further I am not even sure that one array at the time is coherent. Do you at least guarantee this (and if so how considering that you are passing a reference to a native array isn't it?)
We will for now implement your suggestion and re-read all the arrays in the loop until we get the same length for all and assume that it means that we have somewhat coherent information.
Thank you again and we look forward to the reintroduction of some form of transaction mechanism, with the understanding that this is not a trivial thing to implement.

PeterJohnson, 2016-02-05:
The transaction feature was lost in version 2.0 of the protocol.  Individual entries are indeed atomically updated / coherent.  For array entries, the entire array is treated as a single "value", so you'll never get a mix of "old" and "new" values within an array.

If this is a serious problem for you, one option is to pack all of the data into a single entry (most likely of binary type).  GRIP today does not support doing this but it's certainly possible to implement.  When GRIP first talked about doing NT output this was something I suggested they do (partly due to the coherency problem), but they ended up implementing separate NT entries instead.

gartaud, 2016-02-05:
Hi and thank you for jumping in.
It is good to know that an array is treated as a single value.
May I interpret this as meaning that the array returned by double[] getNumberArray(String key, double[] defaultValue) is always a new garbage-collectable array (rather than a reference to a native C/C++ array) and therefore does not need to be copied by the caller)?
Again thank you!

333fred, 2016-02-05:
That is correct, it's a new GC'd array.

gartaud, 2016-02-05:
Great. Thanks!

PS: I am not sure if this is something under your control to but I think that it would be neat to add a note to this page stating that contours reports might not always be formatted as a nXm matrix (and rather should be seen as a collection of independent vectors)
http://wpilib.screenstepslive.com/s/4485/m/50711/l/479908-reading-array-values-published-by-networktables

333fred, 2016-02-05:
@JLLeitschuh, could you look at that?

JLLeitschuh, 2016-02-05:
@ThomasJClark

ThomasJClark, 2016-02-06:
Could we use RPCs?  I'm not completely sure how those work with the new protocol, but it sounds like it would let a robot program register a callback that GRIP can send stuff to when it has everything available.

PeterJohnson, 2016-02-06:
Implementing RPCs for Java is on my TODO list, but haven't been a high priority to date.  I didn't get to it because the "right" way to implement RPCs in Java is via reflection, and I've not taken the time to try to tackle that.  I'm not sure RPC is any better than a packed binary value, as they only work in one direction (client calling server).  My concern is that there are folks writing non-robot programs acting as clients and grabbing GRIP results, and RPCs won't work for them.

JLLeitschuh, 2016-08-10:
Has this feature been added to Java yet?

ThadHouse, 2016-08-10:
I used NetworkTables RPCs for another project recently, and they do not really seem like the right solution for GRIP. They would potentially work if you only had GRIP and the RoboRio, but the way they are implemented the data sent can only be read by the Server. No other client ever receives the RPC info, and there is no way to do Client to Client RPC's. I think we are still looking into implementing arrays of structs, but if we don't get that done, then creating your own data structure using the `raw` data type. The data would always then be synchronized, and any client would be able to access that data.

333fred, 2016-08-10:
Also, see #58 for a bit more discussion on the transaction thing.

gartaud, 2017-03-30:
Hi @333fred 
Any update on this issue?
Thanks!

333fred, 2017-03-30:
I don't believe we're planning on adding transactions. Your best bet is to use an array or use the raw type and encode the info as raw bytes.

virtuald, 2017-05-09:
I think we've agreed that this isn't the right approach, and that for the future struct support is the way to go. Recommend closing this issue in favor of #58.

JLLeitschuh, 2017-06-02:
What about some sort of distributed locking system available per table?
You can grab the lock with some sort of TTL (incase the process doing the locking dies) and then you can write/read safely knowing that anything else that has correctly locked will not be clobbering your data mid-read.

This lock would not be required to be used but if you wanted data consistency guarantees you could use it.

virtuald, 2017-06-03:
@JLLeitschuh that sounds like a bad thing to add to something intended to be used in a realtime environment

PeterJohnson, 2017-09-17:
As mentioned in #58, CBOR and MessagePack are both now available in C++ for storing structured data into NT raw values.  There are various Java libraries for doing the same.  Closing this as we have no plans to implement any NT-level features for transactions--using a raw data type at the NT level with structured data at the application layer is the recommended way to accomplish this.

closed 2017-09-17 by PeterJohnson