wpilibsuite/ntcore #55: Keys never reported as being new in ITableListener

lopsided98, 2016-02-17:
In Java, keys appear to never be reported as being new in a table listener. That is, the "isNew" parameter of the callback is always false. I have not tested other languages.

In the debugging I have done, I cannot find a situation where "isNew" is ever true, but it is possible that the conditions for a key to be considered new have simply changed in a way that I am not aware of since last year.

PeterJohnson, 2016-02-17:
It does work (just tested locally), although it's possible in your test setup that either the value is marked persistent (in which case it will never be "new" because it's loaded at startup), or that you're making the change locally (in which case it won't trigger a listener callback at all, unless you specifically request it with addTableListenerEx()).  My test programs are here: https://gist.github.com/PeterJohnson/c74375e46003602cb097

Output of Listener:
value changed: bar2 isNew=true
value changed: str isNew=true
value changed: numarray isNew=true
value changed: strarray isNew=true
value changed: bar2 isNew=false

Output of ListenerClient:
NT: client: CONNECTED to server 127.0.0.1 port 10000
value changed: foo isNew=true
value changed: bar isNew=true
value changed: foo2 isNew=true
Got foo: 0.5
value changed: str isNew=false
value changed: numarray isNew=false
value changed: strarray isNew=false

PeterJohnson, 2016-02-17:
That being said, the output of ListenerClient, str, numarray, and strarray incorrectly triggered listener callbacks even though they were local changes.  Enabling listening on local changes results in the callback being called twice on each one, once with isNew=true and once with isNew=false.  So there does seem to be a bug here, but it's not with isNew in particular.  I will do some more investigation.

Not notifying on local changes is indeed a change from the previous NetworkTables library.  It was done intentionally to cut down on the number of redundant callbacks / change checks being made.

virtuald, 2016-10-24:
When porting pynetworktables to ntcore, I reused pre-existing listener tests from prior versions of pynetworktables, and I can confirm some weirdness here. I think the culprit in my case is that `storage.notifyEntries` calls new entries with NT_NOTIFY_IMMEDIATE, not NT_NOTIFY_NEW... so these initial calls won't ever be isNew (which was the expectation of prior NT code).

PeterJohnson, 2017-09-17:
Fixed by 77d2b97.

closed 2017-09-17 by PeterJohnson