wpilibsuite/ntcore #56: C++ compiler failed while compiling gtest-all.cc & gmock-all.cc

Dtygbk, 2016-02-18:
Hello, 
I've been trying to build ntcore on Linux, but there are these 2 .cc files that fail to compile.
Output:
```
compiling gtest-all.cc failed.
In file included from /usr/include/c++/5.3.0/random:49:0,
                 from /usr/include/c++/5.3.0/bits/stl_algo.h:66,
                 from /usr/include/c++/5.3.0/algorithm:62,
                 from /home/calvin/ntcore/gmock/gtest/src/gtest.cc:46,
                 from /home/calvin/ntcore/gmock/gtest/src/gtest-all.cc:42:
/usr/include/c++/5.3.0/bits/random.h:106:26: error: expected unqualified-id before ‘__int128’
       { typedef unsigned __int128 type; };
                          ^

compiling gmock-all.cc failed.
In file included from /usr/include/c++/5.3.0/random:49:0,
                 from /usr/include/c++/5.3.0/bits/stl_algo.h:66,
                 from /usr/include/c++/5.3.0/algorithm:62,
                 from /home/calvin/ntcore/gmock/include/gmock/gmock-actions.h:43,
                 from /home/calvin/ntcore/gmock/include/gmock/gmock.h:58,
                 from /home/calvin/ntcore/gmock/src/gmock-all.cc:40:
/usr/include/c++/5.3.0/bits/random.h:106:26: error: expected unqualified-id before ‘__int128’
       { typedef unsigned __int128 type; };
                          ^

```
I've been able to build on Windows, so is there something that I'm missing in GCC?
Also, why is this library not included in WPILib? It seems weird that NetworkTables.jar would be included despite missing a dependency.

PeterJohnson, 2016-02-18:
What platform is this on?  x86, amd64, or something else?  How are you building?  Gradle?  Any flags being set?

I'm confused by what you mean by "not included".  The NetworkTables.jar included with WPILib is of course intended for running on the robot, and only includes the roboRio arm library for size reasons.  There is a separately built "desktop" .jar available that includes libraries for Linux, Mac OS X, and Windows, available at http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilib/networktables/java/NetworkTables/3.0.0-SNAPSHOT/ (look for the most recent -desktop.jar, file size is ~1.5 MB).

Dtygbk, 2016-02-19:
Your link worked! Thanks, my team was searching for for the desktop .jar!

closed 2016-02-19 by 333fred