wpilibsuite/ntcore #72: Add support for adding any number value

SamCarlberg, 2016-06-16:
Right now, calls to `putValue` with a float value (and I suspect `Integer` as well) cause an `IllegalArgumentException` to be thrown.

Lines 722 and 723 in `NetworkTable.putValue(String, Object)` could be changed to the following bolded lines.

<pre>
  @Override
  public boolean putValue(String key, Object value) throws IllegalArgumentException {
    if (value instanceof Boolean)
      return NetworkTablesJNI.putBoolean(path + PATH_SEPARATOR + key, ((Boolean)value).booleanValue());
    <b>else if (value instanceof Number)
      return NetworkTablesJNI.putDouble(path + PATH_SEPARATOR + key, ((Number)value).doubleValue());</b>
    ... // Ommitted for brevity
</pre>

Similarly, `putNumber(String, double)` could be overloaded with a `putNumber(String, Number)` that would just delegate to the first method:
```
public void putNumber(String key, Number value) {
  putNumber(key, value.doubleValue());
}
```

PeterJohnson, 2016-06-16:
Per the discussion in #33 we decided not to add support for Number, but I can see the argument for supporting Double in putValue.  Or is the issue with putValue that putValue(1) results in an Integer object being passed, which isn't an instance of Double and so would result in an IllegalArgumentException?

SamCarlberg, 2016-06-17:
It's mostly to do with `putValue`. I suggested a `putNumber` overload to offer a more explicit function, but I don't think it needs to be added.

Yes, `putValue("key", 1)` will throw an `IllegalArgumentException`.

The issue with it is that it only checks for `value instanceof Double`. It doesn't check for `Integer` (or any smaller integer type like `Short` or `Byte`), nor does it check for `Float`, which can be safely cast to `Double` without losing precision. (`putNumber(key, float)` *does* work, though, due to the compiler promoting `float` to `double`). The safest way would be to change the check from
```
if (value instanceof Double)
```
to
```
if (value instanceof Byte || value instanceof Character || value instanceof Short || value instanceof Integer || value instanceof Float || value instanceof Double)
```

Since that only excludes `Long` (and `BigInteger`, which I doubt would be used by a user), that big `if` could be
```
if (value instanceof Number && !(value instanceof Long))
```

Additionally, the exception message could be better. Something like:
```
throw new IllegalArgumentException("Value of type " + value.getClass().getName() + " cannot be put into a table");
```

333fred, 2016-06-17:
I'd be more in favor of changing the internal changes to support that case, rather than external facing changes. To my mind, this is a bug in `putValue`, but fixing it shouldn't expose a new public function. It would still fail if I had a number as an object, like this: `putValue((Object)  1)`.

SamCarlberg, 2016-06-17:
@333fred `putValue("key", (Object) 1)` *should* work; the cast to `Object` won't hide the runtime type of the value (which will still be `Integer`). It won't though, because `putValue` only checks for `instanceof Double`

See https://ideone.com/KEkWqc for an example.

333fred, 2016-06-17:
Here, I think this version more illustrates my point: http://ideone.com/kc1mdS. Note that even though there is a version of `putValue` for `Number`, Java calls the `Object` version. This is because, at compile-time, Java doesn't know the type of the argument to be `Integer`, it only knows it to be `Object`. Now, `putValue("key", (Object) 1)` is a trivial example of this that you would never do, but there are certainly situations in which you could have a value for a key and not know its type, and therefore call `putValue` with nothing more specific than an object.

SamCarlberg, 2016-06-17:
There isn't `putValue(String, Number)` overload in `ITable`, though, and didn't you argue that we shouldn't have one?

333fred, 2016-06-17:
That is my argument. My example shows how just adding the `Number` overload isn't enough, so we really just need to fix a bug in `putValue`.

SamCarlberg, 2016-06-17:
Right, I was saying we should `putValue` to accept values that are already valid with `putNumber`, and maybe adding an overload with `Number`, but I don't think that would be needed.

virtuald, 2016-10-18:
FWIW, pynetworktables has historically allowed float and integers to be passed to `putValue`. This will continue to be the case in the rewritten ntcore-based pynetworktables.

commit d977c067 (2016-10-20): Java: Allow any Number type other than Long to be passed to putValue().

Also improve exception message.

Fixes #72.
commit a8eb9080 (2016-10-21): Java: Allow any Number type (not just Double) to be passed to putValue().

Also improve exception message.

Fixes #72.
PR #129 (2016-10-22): Java: Allow any Number type other than Long to be passed to putValue().
Also improve exception message.

Fixes #72.
commit 011ac1fa (2016-10-22): Java: Allow any Number type (not just Double) to be passed to putValue(). (#129)

Also improve exception message.

Fixes #72.
closed 2016-10-22 by PeterJohnson