wpilibsuite/ntcore #77: 111 Errors

balloman, 2016-06-27:
I am working with VS2015 and I am sure I have done everything write, as I have no errors in my code, however when I build I get this:
```
Severity	Code	Description	Project	File	Line
Error	LNK1169	one or more multiply defined symbols found	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\x64\Debug\RocketTracker.exe	1
Error	LNK2005	_Getcvt already defined in msvcprtd.lib(MSVCP140D.dll)	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(xwctomb.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(ios.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(iosptrs.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(locale.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(locale0.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(StlCompareStringA.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(StlCompareStringW.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(StlLCMapStringA.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(StlLCMapStringW.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(winapinls.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(winapisupp.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(wlocale.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(xdateord.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(xlocale.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(xlock.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(xthrow.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(Base64.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(Dispatcher.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(leb128.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(Log.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(Message.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(NetworkConnection.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(NetworkTable.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(Notifier.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(ntcore_cpp.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(raw_istream.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(raw_socket_istream.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(RpcServer.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(SafeThread.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(SmallVector.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(SocketError.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(Storage.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(StringMap.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(StringRef.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(TableKeyNotDefinedException.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(TCPAcceptor.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(TCPConnector.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(TCPStream.obj)	1
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Source.obj	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\ntcore.lib(timestamp.obj)	1
[... cut ...]
Error	LNK2005	"private: static class std::locale::_Locimp * __cdecl std::locale::_Init(bool)" (?_Init@locale@std@@CAPEAV_Locimp@12@_N@Z) already defined in msvcprtd.lib(MSVCP140D.dll)	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(locale0.obj)	1
Error	LNK2005	"private: static class std::locale::_Locimp * __cdecl std::locale::_Getgloballocale(void)" (?_Getgloballocale@locale@std@@CAPEAV_Locimp@12@XZ) already defined in msvcprtd.lib(MSVCP140D.dll)	RocketTracker	C:\Users\benar\Desktop\FRC\RocketTracker\RocketTracker\libcpmt.lib(locale0.obj)	1

```

PeterJohnson, 2016-06-27:
The errors you're getting are because you're building with different compiler options (release/debug, secure iterators) than what the library was built with.  See https://github.com/wpilibsuite/ntcore/blob/master/toolchains/windows.gradle for the compiler flags used for ntcore.  If you build ntcore.lib locally you can tweak this file to change the compiler options to match your project.  If you downloaded ntcore.lib, it's also possible that the one you downloaded was built with a different version of MSVC, so compiling it yourself would also fix any issues that might be caused by that.

See also http://stackoverflow.com/questions/14714877/mismatch-detected-for-runtimelibrary and http://stackoverflow.com/questions/7668200/error-lnk2038-mismatch-detected-for-iterator-debug-level-value-0-doesnt

balloman, 2016-06-30:
How do i go about changing these compiler options then?

erikuhlmann, 2016-07-03:
I just did this.

Check this flag on your project properties:

[image]

Then edit the following line (18) of `toolchains/windows.gradle`
```gradle
args << '/EHsc' << '/DNOMINMAX' << '/D_SCL_SECURE_NO_WARNINGS' << '/D_WINSOCK_DEPRECATED_NO_WARNINGS'
```
and add this to the end (replace `/MDd` with whatever option was on the properties page)
```gradle
<< '/MDd'
```
I also added the following to compile it as debug
```gradle
<< '/Zi' << '/Od' << '/FS'
```

Then delete the `native` folder. If you try recompiling now then linking tests will fail with the same error as you have in your project, so instead run gradle with `gradlew :native:build -PwithoutTests`

balloman, 2016-07-04:
All went well until I got this error when building:

```
Severity	Code	Description	Project	File	Line
Error	C4996	'ITable::GetBoolean': Raises an exception if key not found; use GetBoolean(StringRef key, bool defaultValue) instead	RocketTracker	c:\users\benar\onedrive\documents\programming\ntcore\include\networktables\networktable.h	352
Error	C4996	'ITable::GetNumber': Raises an exception if key not found; use GetNumber(StringRef key, double defaultValue) instead	RocketTracker	c:\users\benar\onedrive\documents\programming\ntcore\include\networktables\networktable.h	283
Error	C4996	'ITable::GetString': Raises an exception if key not found; use GetString(StringRef key, StringRef defaultValue) instead	RocketTracker	c:\users\benar\onedrive\documents\programming\ntcore\include\networktables\networktable.h	317

```

ThadHouse, 2016-07-04:
The exception raising functions are depreciated. You either need to turn off the error for C4996 in Visual Studio, or change the functions that use that call to use the functions that take a defaultValue parameter instead.

erikuhlmann, 2016-07-04:
The annoying thing is those seem to generate errors even when those overloads of the methods aren't being called in your code. I added this define before including ntcore headers

```
#define NT_DEPRECATED(msg)
```

balloman, 2016-07-07:
Thank you guys, I got my info from c++ to labview finally. Thanks for all the help

closed 2016-07-07 by balloman

ThadHouse, 2016-07-07:
We're there specific parts of the C api missing that LabVIEW needed. I would be interested in seeing what was and potentially adding them to ntcores itself.

balloman, 2016-07-07:
Well, I just used network tables and retrieved the value in labview

ThadHouse, 2016-07-07:
Ahh I was thinking you were writing a wrapper for LabVIEW. Nevermind