wpilibsuite/ntcore #184: SetUpdateRate inconsistencies

gbear605, 2017-01-18:
The documentation for setUpdateRate is inconsistent. In the end location of it (Dispatcher.cpp), it says

  ```
// don't allow update rates faster than 10 ms or slower than 1 second
  if (interval < 0.01)
    interval = 0.01;
  else if (interval > 1.0)
    interval = 1.0;
```

But in NetworkTable.h, it says "update interval in seconds (range 0.1 to 1.0)" in the documentation comment. It says something similar in the Javadoc.

In addition, though this could be an implementation issue on my part, when I tried to use setUpdateRate with 0.01, it appeared to be at 100ms instead of 10ms.

PeterJohnson, 2017-01-19:
You're right about the docs; will fix.  Note the update rate timing only affects the local NT update rate; it has no effect on other nodes in the network.  E.g. if you set it on the robot, the robot will send updates every 10 ms, but a connected dashboard will still send its updates at 100 ms.

closed 2017-01-19 by PeterJohnson

gbear605, 2017-01-19:
Ah, that's the reason why it didn't appear to work for me. Thanks for the
help.

Also, if you don't mind answering, what's the design reason for limiting it
at 10ms minimum? Is it because of bandwidth? It doesn't appear to be in the
standards document for the protocol.

PeterJohnson, 2017-01-19:
The reason to limit it is airtime on the field, yes.  Many small packets can swamp the available wireless time even if the actual bandwidth utilization isn't that high.  However, we did extensive testing that showed 10 ms didn't have much more impact than 100 ms.  There should not really be a reason to go faster than that in any case.  No exact value is mentioned in https://github.com/wpilibsuite/ntcore/blob/master/doc/networktables3.adoc#bandwidth but the consideration is.

gbear605, 2017-01-19:
Thanks for the help.

commit 13457d1b (2017-01-19): Update setUpdateRate() documentation to match implementation.

Update rates as fast as 10 ms are now allowed.

Fixes #184.