wpilibsuite/ntcore #185: Building on Raspberry Pi

Bubba74, 2017-01-21:
I am trying to build this repo on my Raspberry Pi, but am having some issues.

Cloning the repo worked fine, and even building it with './gradlew :native:build' worked fine. The problem is that despite gradle saying the build succeeded, I cannot find the libntcore.a library anywhere on the RPi.

Running './gradlew build' does not work (throwing an error with the g++ -m32 arg) but I don't think it worked on a Mac either.

Any advice is much appreciated,
         Henry Loh

ThadHouse, 2017-01-21:
The right way to build on a pi is to run

`./gradlew :arm:ntcore:build -PcompilerPrefix=` 

(yes there is nothing after the equals statement.

After this, the builds will be in arm/ntcore/build/libs.

However we publish pi artifacts to our maven repo which are already built for the pi. Use the arm-raspbian classifier. Also note you will need wpiutil if you want to link to the static archives.

Bubba74, 2017-01-21:
Thanks for getting back to me so quickly, I ran what you suggested and I imagine it would work but the RPi froze for the last 20ish minutes.
I'm planning on unplugging the RPi and then getting the builds from the maven repo you mentioned. Does that sound right?

ThadHouse, 2017-01-21:
It probably didn't freeze, it's probably just going to take over an hour to compile natively on a pi. But yeah the maven artifacts are probably an easier solution.

Bubba74, 2017-01-21:
Yes! It works! Thank you so much for your help, I really appreciate it.

closed 2017-01-21 by Bubba74