wpilibsuite/ntcore #186: NT: ERROR: could not resolve address(TCPConnectorr)

balloman, 2017-01-25:
I keep getting this error when using the SetTeam function, how do I fix it?

PeterJohnson, 2017-01-27:
As SetTeam() configures 4 or so addresses and round robins between them, you will nearly always get this error (really it shouldn't be an error but rather seen as informational).  You can safely ignore it as long as you eventually get a CONNECTED message (which signals it has successfully connected to the server).  If you see an endless loop of these errors and never get the CONNECTED message that means it can't find the server at all and you should check your network configuration, that the server is actually running, etc.

balloman, 2017-01-27:
Ah, I get it in a loop. I'm not sure what's wrong our team has been working for days to figure out why we can't connect to the robot.

balloman, 2017-01-27:
Every time I get a time out error

PeterJohnson, 2017-01-27:
This can be due to a lot of different reasons.  Some things to check:
- Is robot code running?  (e.g. does the driver station show that it's connected with a teleoperated enabled/disabled message?)
- Do you have any firewall software running on the computer that might be blocking outgoing connections?
- Can the LabView dashboard connect? (it uses NetworkTables as well)

balloman, 2017-01-27:
Yes

No

Yes

PeterJohnson, 2017-01-27:
Can you open a command window and successfully ping "roborio-xxx-frc.local" or "roborio-xxx-frc.lan" (where xxx is your team number)?  Are you connected via Ethernet or USB?

balloman, 2017-01-27:
Connected wirelessly, if I ping it I get a response back at 10.0.34.1(team 34) and a message saying port unreachable

PeterJohnson, 2017-01-27:
The DS and the NetworkTables application are running on the same computer, right?

balloman, 2017-01-27:
Yes

PeterJohnson, 2017-01-27:
If you restart the DS application can it still connect?  Which address resolved, the .local address or the .lan address? (should show up as "Pinging roborio-xxx-frc.local [10.0.34.xxx]"). Can you ping 10.0.34.2 successfully?

PeterJohnson, 2017-01-27:
Do you ever see the message "client: DS overriding server IP to xxxx"?  If so, what is xxxx?  If you have PuTTY, can you try opening a raw connection to localhost port 1742 and see what it prints?

PeterJohnson, 2017-01-27:
What is your Driver Station version number?

balloman, 2017-01-28:
If I ping 10.0.34.2 I get one reply from 10.0.34.138 with Destination host unreachable, and the remaining three is a reply from 10.0.34.1 with destination port unreachable
1. I did not see that message
2. I can still restart the ds and have it connect.
3. If i try to connect to either the robot or the local host i get connection refused.
4. 17.0a11

PeterJohnson, 2017-01-28:
I really don't know what's going on.  If you have DS 17.0a11 running on your local computer, and run PuTTY, select connection type "Raw", port "1742", and host "localhost", you should *definitely* get a connection and some text back.

closed 2017-02-09 by balloman