wpilibsuite/ntcore #188: Access Violation

balloman, 2017-01-28:
Whenever I call any function from the Network Tables library, I get this Access Exception error. Why is this happening, and how can I fix it.
`Unhandled exception at 0x00A90000 in NTTest.exe: 0xC0000005: Access violation executing location 0x00A90000. occurred`

bradamiller, 2017-01-28:
Can you post the zip of a small project that does that?

balloman, 2017-01-28:
`// NTTest.cpp : Defines the entry point for the console application.
//

#include<iostream>
#include "stdafx.h"
#include "ntcore.h"
#include "networktables\NetworkTable.h"
#include <string>

using std::shared_ptr;

shared_ptr<NetworkTable> startConnection() {
	NetworkTable::SetClientMode();
	NetworkTable::SetIPAddress("localhost");
	shared_ptr<NetworkTable> myTable = NetworkTable::GetTable("SmartDashboard");
	return myTable;
}

int main()
{
	shared_ptr<NetworkTable> table = startConnection();
    return 0;
}
`
There you go.

bradamiller, 2017-01-28:
Didn't realize it was a PC program. Can you let us know what version of Visual Studio your are running and where you got the network tables libraries.

balloman, 2017-01-28:
I had a mentor fix it, he mentioned something about the ntcore code never exporting the networktables or something.

closed 2017-01-28 by balloman