wpilibsuite/ntcore #190: Need docs for table notification keys

SamCarlberg, 2017-02-25:
Currently, the flag constants are just declared like this (Java):

```java
  /** Notifier flag values. */
  public static final int NOTIFY_IMMEDIATE = 0x01;
  public static final int NOTIFY_LOCAL = 0x02;
  public static final int NOTIFY_NEW = 0x04;
  public static final int NOTIFY_DELETE = 0x08;
  public static final int NOTIFY_UPDATE = 0x10;
  public static final int NOTIFY_FLAGS = 0x20;
```

This generates documentation for `NOTIFY_IMMEDIATE` that just says "Notifier flag values', and no documentation at all for the rest.

ThadHouse, 2017-02-25:
If someone is looking for docs for now before we can get them implemented here, my C# docs are located at https://robotdotnet.github.io/examples/nt/tablelisteners.html#using-the-notifyflags-parameter which has documentation on each flag.

PeterJohnson, 2017-09-07:
Fixed by #204.

closed 2017-09-07 by PeterJohnson