wpilibsuite/ntcore #221: wait for notifier queue to clear

nightpool, 2017-08-09:
On Shuffleboard, we [currently have some complex logic](https://github.com/wpilibsuite/shuffleboard/pull/118) that we use in testing to determine when the callbacks have already fired. This allows us to assert things about the state of the application after adding keys, changing values, etc. 

However, this would be much more cleanly implemented at the network table level as either a blocking call or a Future that we can join to wait until the queue has been drained. In testing environments specifically, many libraries with asynchronous processing provide synchronous versions of their queue processing code, to ensure consistency and repeatability across tests. Then the flow would be "make network table update -> call `drainQueue()` -> assert". I don't know whether that would be feasible for ntcore to implement with it's current architecture, but it's something to look into.

So the minimum feature here is a blocking call or Future to wait on until the queue has cleared. The idea of a synchronous processing mode for testing is a possible extension of that, but right now the biggest need is some way of dealing with network tables in unit tests at all.

PeterJohnson, 2017-08-09:
To clarify, you want a function that blocks until the notifier queue is empty?  I'm assuming you want it to return immediately if the queue is already empty?  Note this will only be effective for local notifications if you've created at least one NOTIFY_LOCAL listener.

AustinShalit, 2017-08-09:
Yes.  That feature would be great!

nightpool, 2017-08-09:
> Note this will only be effective for local notifications if you've created at least one NOTIFY_LOCAL listener

can you expand on this a little more? What do you mean by that?

JLLeitschuh, 2017-08-10:
Would `drainQueue()` return a `CompletableFuture<Void>` or would `drainQueue()` be blocking?
If it is blocking then `drainQueue()` should have another implementation that takes a timeout.

PeterJohnson, 2017-08-10:
For efficiency reasons, local changes are not queued unless at least one local listener has been created.

PeterJohnson, 2017-08-10:
I'm not liking drainQueue as a name.  WaitForEmpty is a better name for what it's actually doing.  Remember this has to be done at the C++ level, so it will be blocking.  I can provide a timeout parameter.

PeterJohnson, 2017-08-10:
Alternatively, similar to RPCs, I could provide a non-callback interface for notification processing.  E.g. so you'd call a blocking GetNotification() function in a loop... this would be a bit more work to implement but would also let us move the notification processing to the Java level rather than having C++ call back through JNI...

nightpool, 2017-08-10:
again, there are two proposals here:

 - `waitForQueue()` a blocking or future-based method that allows consumers to wait until the queue is empty

 - `drainQueue()` and some way (`setQueueMode()`?) to allow consumers to decide *when* the queue is handled and always enforce single-threaded, deterministic queue processing.

Also, just because the c++ interface is blocking doesn't mean the java version can't return a future.

PeterJohnson, 2017-09-04:
I think this can be done with a single waitForQueue method with a timeout (timeout=0 is non-blocking, timeout=-1 is wait forever), combined with the new polling feature (which pulls events directly from the queue).  For deterministic processing, you waitForQueue, then call Poll with timeout=0 to get the events that were flushed out of the queue.

PeterJohnson, 2017-09-07:
Fixed by #204.

closed 2017-09-07 by PeterJohnson