wpilibsuite/ntcore #231: Cannot build with cmake anymore

rtrahms, 2017-09-16:
CMakelists.txt is missing.  Do you have pointers on how to now build ntcore with cmake?
Thanks!

rtrahms, 2017-09-16:
My build info: Ubuntu 16.04 (Jetson TX1)

PeterJohnson, 2017-09-16:
Is there a reason that gradle does not work for you?  One of the reasons we removed cmake was that we were not regularly testing that build approach.  The other is that we split off the wpiutil library to a separate repository, so dependency management is now a concern.

rtrahms, 2017-09-16:
Yes, I have tried gradle, and so far have been unsuccessful.  I know the straight ./gradlew build fails, but I have also tried ./gradlew :native:build and ./gradlew :arm:build.  The latter two complain that a Project 'native' or Project 'arm' not found in root project.

PeterJohnson, 2017-09-16:
The readme is out of date.  As I don't think we've published TX1 artifacts yet for wpiutil, first you'll need to clone, build, and locally publish the wpiutil repo (https://github.com/wpilibsuite/wpiutil) with `./gradlew -PskipAthena publish`.  Then build ntcore with `./gradlew -PskipAthena build`.

rtrahms, 2017-09-16:
I'll give that a try - thanks.

rtrahms, 2017-09-16:
I was able to successfully clone, build and publish wpiutil as above.  When I attempted the ntcore build, I received a different error this time...

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'ntcore'.
> Exception thrown while executing model rule: DependencyConfigRules#setupDependencyDownloads(ModelMap<Task>, DependencyConfigSpec, BinaryContainer, ProjectLayout, BuildConfigSpec)
   > Could not resolve all files for configuration ':nativeDeps'.
      > Could not find wpiutil-cpp-linuxarm.zip (edu.wpi.first.wpiutil:wpiutil-cpp:424242.2.0.0-20170916085141-34-g3faecdb).
        Searched in the following locations:
            file:/home/ubuntu/releases/maven/development/edu/wpi/first/wpiutil/wpiutil-cpp/424242.2.0.0-20170916085141-34-g3faecdb/wpiutil-cpp-424242.2.0.0-20170916085141-34-g3faecdb-linuxarm.zip

ThadHouse, 2017-09-17:
What is the folder structure looking like at home/ubuntu/releases/maven/development/edu/wpi/first/wpiutil? It should have published right. also try running ntcore with --refresh-dependencies as an extra parameter.

rtrahms, 2017-09-17:
The structure for wpiutil has wpiutil-cpp and wpiutil-java folders. Each subfolder has a set of maven-metadata files and subfolders.

Adding --refresh-dependences to the ./gradlew -PskipAthena build  in ntcore did not change anything.

PeterJohnson, 2017-09-17:
What files are in `/home/ubuntu/releases/maven/development/edu/wpi/first/wpiutil/wpiutil-cpp/424242.2.0.0-20170916085141-34-g3faecdb/`?  Or does that directory not exist?

rtrahms, 2017-09-17:
One further clue - I went looking for the file in that structure that caused the failure.  There is a zip file in that location, but it is named slightly differently:

wpiutil-cpp-424242.2.0.0-20170916085141-34-g3faecdb-linuxarm-v7.zip

Is the v7 at the end causing a problem?

PeterJohnson, 2017-09-17:
Short answer: yes.  You might be able to work around the issue by renaming the .zip (and the .zip.md5 / .zip.sha1 files) to remove the -v7 until we figure out what's going on.

rtrahms, 2017-09-17:
Yes it does!  I removed the "-v7" and the build proceeded further.  Still crashed, but on a different issue:

ubuntu@tegra-ubuntu:~/git-repos/ntcore$ ./gradlew -PskipAthena build
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
:cppHeadersZip
:compileJava
Download http://first.wpi.edu/FRC/roborio/maven/development/edu/wpi/first/wpiutil/wpiutil-java/maven-metadata.xml
Download http://first.wpi.edu/FRC/roborio/maven/development/edu/wpi/first/wpiutil/wpiutil-java/2.0.0-20170910202423-34-g3faecdb/wpiutil-java-2.0.0-20170910202423-34-g3faecdb.pom
Note: /home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/wpilibj/networktables/NetworkTable.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
:processResources NO-SOURCE
:classes
:ntcoreJNIjniHeaders
:ntcorejniHeaders
:cppSourcesZip
:jar
:javadoc/home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/networktables/NetworkTable.java:263: warning - @inheritDoc used but getPath() does not override or implement any method.
/home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/networktables/NetworkTable.java:263: warning - @inheritDoc used but getPath() does not override or implement any method.
/home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/networktables/NetworkTable.java:263: warning - @inheritDoc used but getPath() does not override or implement any method.

3 warnings
:javadocJar
:downloadedu.wpi.first.wpiutilwpiutil-cppheaders
:compileNtcoreDevExecutableNtcoreDevCpp
:compileNtcoreSharedLibraryNtcoreCpp/home/ubuntu/git-repos/ntcore/src/main/native/cpp/jni/NetworkTablesJNI.cpp:1:17: fatal error: jni.h: No such file or directory
compilation terminated.

 FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileNtcoreSharedLibraryNtcoreCpp'.
> A build operation failed.
      C++ compiler failed while compiling NetworkTablesJNI.cpp.
  See the complete log at: file:///home/ubuntu/git-repos/ntcore/build/tmp/compileNtcoreSharedLibraryNtcoreCpp/output.txt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 44s
11 actionable tasks: 11 executed

PeterJohnson, 2017-09-17:
Ok, something else for us to look at.  The workaround for that issue will be to edit build.gradle and add `'src/arm-linux-jni'` to the exportedHeaders srcDirs lists on lines 157 and 171 (so they read `srcDirs 'src/main/native/include', 'src/arm-linux-jni'`.  You might also need to add `'src/arm-linux-jni/linux'` to the list as well.

PeterJohnson, 2017-09-17:
To help us fix this issue on our side, what are the contents of `build/tmp/compileNtcoreJNISharedLibraryNtcoreJNICpp/options.txt`?

rtrahms, 2017-09-17:
Progress!  I added both include paths you mentioned above.  The build proceeds further - now looking for a different include file...

ubuntu@tegra-ubuntu:~/git-repos/ntcore$ ./gradlew -PskipAthena build
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
:cppHeadersZip UP-TO-DATE
:compileJava UP-TO-DATE
:processResources NO-SOURCE
:classes UP-TO-DATE
:ntcoreJNIjniHeaders UP-TO-DATE
:ntcorejniHeaders UP-TO-DATE
:cppSourcesZip UP-TO-DATE
:jar UP-TO-DATE
:javadoc UP-TO-DATE
:javadocJar UP-TO-DATE
:downloadedu.wpi.first.wpiutilwpiutil-cppheaders UP-TO-DATE
:compileNtcoreDevExecutableNtcoreDevCpp
:compileNtcoreSharedLibraryNtcoreCpp/home/ubuntu/git-repos/ntcore/src/main/native/cpp/jni/NetworkTablesJNI.cpp:4:58: fatal error: edu_wpi_first_networktables_NetworkTablesJNI.h: No such file or directory
compilation terminated.

 FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileNtcoreSharedLibraryNtcoreCpp'.
> A build operation failed.
      C++ compiler failed while compiling NetworkTablesJNI.cpp.
  See the complete log at: file:///home/ubuntu/git-repos/ntcore/build/tmp/compileNtcoreSharedLibraryNtcoreCpp/output.txt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 34s
11 actionable tasks: 2 executed, 9 up-to-date

ThadHouse, 2017-09-17:
Do you have the JDK installed? Not just a JRE? The headers task requires a full JDK, not just the JRE.

ThadHouse, 2017-09-17:
If you do, try adding `--rerun-tasks` to the list of gradle command line flags. That should force the JNI task to rerun

rtrahms, 2017-09-17:
@Thad -
Yes, I have both JDK and JRE installed.

I'll try the --rerun-tasks in a moment.

rtrahms, 2017-09-17:
Same result, even with --rerun-tasks...

:compileNtcoreSharedLibraryNtcoreCpp/home/ubuntu/git-repos/ntcore/src/main/native/cpp/jni/NetworkTablesJNI.cpp:4:58: fatal error: edu_wpi_first_networktables_NetworkTablesJNI.h: No such file or directory
compilation terminated.

 FAILED

ThadHouse, 2017-09-17:
What are the contents of `build/tmp/compileNtcoreSharedLibraryNtcoreCpp/options.txt`? Also what files are in `build/ntcoreJNI`?

rtrahms, 2017-09-17:
Here are the contents of options.txt in that directory:

-x
c++
-c
-fPIC
-I
/home/ubuntu/git-repos/ntcore/src/main/native/include
-I
/home/ubuntu/git-repos/ntcore/src/arm-linux-jni
-I
/home/ubuntu/git-repos/ntcore/src/arm-linux-jni/linux
-I
/home/ubuntu/git-repos/ntcore/build/dependencies/wpiutil-cpp/headers
-std=c++11
-Wformat=2
-Wall
-Wextra
-Werror
-pedantic
-Wno-psabi
-g
-Wno-unused-parameter
-fPIC
-rdynamic
-Wno-error=deprecated-declarations
-pthread

The contents of the ntcore directory is another directory named jnlinclude.  In that directory there is a single file named edu_wpi_first_networktables_NetworkTablesJNI.h

ThadHouse, 2017-09-17:
Interesting. There should be another include directory. Add `/home/ubuntu/git-repos/ntcore/build/ntcore/jniinclude` to the exportedHeaders srcDirs array you modified earlier. I need to get my hands on a TX1 to test some of this stuff.

rtrahms, 2017-09-17:
Closer!  I modified lines 157 and 171 in the build.gradle file with the directory above.  It's now moving beyond that to another error (linker error?)...

ubuntu@tegra-ubuntu:~/git-repos/ntcore$ ./gradlew --rerun-tasks build
Starting a Gradle Daemon (subsequent builds will be faster)
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
:cppHeadersZip
:compileJavaNote: /home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/wpilibj/networktables/NetworkTable.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

:processResources NO-SOURCE
:classes
:ntcoreJNIjniHeaders
:ntcorejniHeaders
:cppSourcesZip
:jar
:javadoc/home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/networktables/NetworkTable.java:263: warning - @inheritDoc used but getPath() does not override or implement any method.
/home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/networktables/NetworkTable.java:263: warning - @inheritDoc used but getPath() does not override or implement any method.
/home/ubuntu/git-repos/ntcore/src/main/java/edu/wpi/first/networktables/NetworkTable.java:263: warning - @inheritDoc used but getPath() does not override or implement any method.

3 warnings
:javadocJar
:downloadedu.wpi.first.wpiutilwpiutil-cppheaders
:compileNtcoreDevExecutableNtcoreDevCpp
:compileNtcoreSharedLibraryNtcoreCpp
:linkNtcoreSharedLibrary
:ntcoreSharedLibrary
:linkNtcoreDevExecutable/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::operator<<(void const*)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::StringRef::find_first_not_of(llvm::StringRef, unsigned long) const'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::convertUTF8ToUTF16String(llvm::StringRef, llvm::SmallVectorImpl<unsigned short>&)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::operator<<(long)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::StringMapImpl::FindKey(llvm::StringRef) const'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `vtable for wpi::raw_socket_istream'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::operator<<(llvm::format_object_base const&)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `vtable for llvm::raw_svector_ostream'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `wpi::Base64Decode(llvm::StringRef, unsigned long*, llvm::SmallVectorImpl<char>&)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::write(unsigned char)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `wpi::raw_fd_istream::~raw_fd_istream()'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::~raw_ostream()'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::StringMapImpl::RemoveKey(llvm::StringMapEntryBase*)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::errs()'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_string_ostream::~raw_string_ostream()'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::handle()'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::StringRef::c_str(llvm::SmallVectorImpl<char>&) const'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::format_object_base::home()'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `vtable for llvm::format_object_base'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `vtable for wpi::raw_mem_istream'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::StringRef::find(llvm::StringRef, unsigned long) const'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::write(char const*, unsigned long)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_fd_ostream::~raw_fd_ostream()'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::getAsUnsignedInteger(llvm::StringRef, unsigned int, unsigned long long&)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::operator<<(unsigned long)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::StringMapImpl::RehashTable(unsigned int)'
/home/ubuntu/git-repos/ntcore/build/libs/ntcore/shared/libntcore.so: undefined reference to `llvm::raw_ostream::preferred_buffer_size() const'
[... cut ...]
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':linkNtcoreDevExecutable'.
> A build operation failed.
      Linker failed while linking ntcoreDev.
  See the complete log at: file:///home/ubuntu/git-repos/ntcore/build/tmp/linkNtcoreDevExecutable/output.txt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 3s
13 actionable tasks: 13 executed

PeterJohnson, 2017-09-18:
It's not linking to wpiutil.so when building the dev executable for some reason.  Most likely due to some other place gradle is injecting the v7 nomenclature?  It successfully built ntcore.so however... if you're going to use this with Java, you might try running just `./gradlew test` which will only build and run the java tests.

ThadHouse, 2017-09-18:
That would actually explain a few other issues as well. Try 1 more thing. In config.gradle, change line 215 to `nativearm`.  Make the same change in wpiutil as well, then rebuild and republish wpiutil. Then attempt an ntcore build again.

rtrahms, 2017-09-18:
I should have mentioned this - my preferred implementation is in c, not java.  I can do both of course, but I have previously built and used ntcore successfully on Raspbian with c (using cmake).   I will try the steps above today.

PeterJohnson, 2017-09-18:
In that case, the .so was successfully built... so you can go ahead and use it (you just need to grab the built ntcore.so and wpiutil.so).

rtrahms, 2017-09-19:
I grabbed wputil.so and ntcore.so and was able to link them to complete my g++ build.
Thanks for the help!

PeterJohnson, 2018-05-02:
wpilibsuite/allwpilib#1015 will bring back cmake builds.

closed 2018-05-02 by PeterJohnson