wpilibsuite/ntcore #236: Error when Gradle Building ntcore

JDaniel29, 2017-09-30:
Hello everyone,

My team is currently trying to build the ntcore release for our Jetson TK1. However, whenever we run the `./gradlew build` command, we get the error below.

```
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'ntcore'.
> Exception thrown while executing model rule: DependencyConfigRules#setupDependencyDownloads(ModelMap<Task>, DependencyConfigSpec, BinaryContainer, ProjectLayout, BuildConfigSpec)
   > Could not resolve all files for configuration ':nativeDeps'.
      > Could not find wpiutil-cpp-linuxarm.zip (edu.wpi.first.wpiutil:wpiutil-cpp:424242.3.0.0-alpha-1-20170930140408-1-g9fdb33b).
        Searched in the following locations:
            file:/home/ubuntu/releases/maven/development/edu/wpi/first/wpiutil/wpiutil-cpp/424242.3.0.0-alpha-1-20170930140408-1-g9fdb33b/wpiutil-cpp-424242.3.0.0-alpha-1-20170930140408-1-g9fdb33b-linuxarm.zip

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
```

We have tried following the advice in [this issue](https://github.com/wpilibsuite/ntcore/issues/231) and while we could build the wpiutil repository, we still can't build ntcore. What should we try to get this working?

Thanks,

Jonathan Daniel

PeterJohnson, 2017-09-30:
#237 should fix this.

JDaniel29, 2017-09-30:
Thank you for your help. When I ran this, I now have a different error.

```
Download https://repo1.maven.org/maven2/org/hamcrest/hamcrest-core/1.3/hamcrest-core-1.3.jar
:processTestResources NO-SOURCE
:testClasses
:test

edu.wpi.first.networktables.JNITest > jniLinkTest FAILED
    java.lang.UnsatisfiedLinkError at JNITest.java:9

edu.wpi.first.networktables.ConnectionListenerTest > testThreaded FAILED
    java.lang.NoClassDefFoundError at ConnectionListenerTest.java:14

edu.wpi.first.networktables.ConnectionListenerTest > testJNI FAILED
    java.lang.NoClassDefFoundError at ConnectionListenerTest.java:14

edu.wpi.first.networktables.LoggerTest > testLogger FAILED
    java.lang.NoClassDefFoundError at LoggerTest.java:13

edu.wpi.first.networktables.EntryListenerTest > testPrefixNewRemote FAILED
    java.lang.NoClassDefFoundError at EntryListenerTest.java:14

5 tests completed, 5 failed
:test FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':test'.
> There were failing tests. See the report at: file:///home/ubuntu/backup-ntcore/ntcore/build/reports/tests/test/index.html

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 31s
31 actionable tasks: 11 executed, 20 up-to-date
```

PeterJohnson, 2017-09-30:
The issue is the JNI isn't loading (the JNITest.java:9 failure).  What does `unzip -l build/ntcore-nativeTestFiles.jar` show (or `find build -name \*.so`)?  I'm trying to find out how to get more detail on the UnsatisfiedLinkError to find out what path it's looking for, but these two items will help debug this issue.

PeterJohnson, 2017-09-30:
Ah, also run `./gradlew run` and provide that output (in particular, the line after "Hello World!").  Thanks.

JDaniel29, 2017-09-30:
So, `unzip -1 build/ntcore-nativeTestFiles.jar` shows this: 
```
Archive:  build/ntcore-nativeTestFiles.jar
  Length      Date    Time    Name
---------  ---------- -----   ----
        0  2017-09-30 15:03   META-INF/
       25  2017-09-30 15:03   META-INF/MANIFEST.MF
        0  2017-09-30 15:03   linux/
        0  2017-09-30 15:03   linux/nativearm/
 10987830  2017-09-30 14:53   linux/nativearm/libntcore.so
---------                     -------
 10987855                     5 files
```

And `./gradlew run` shows
```
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
Currently building a debug binary. To build release, add -PreleaseBuild to your gradle command.
:compileJava UP-TO-DATE
:processResources NO-SOURCE
:classes UP-TO-DATE
:compileDevJava
:processDevResources NO-SOURCE
:devClasses
:downloadedu.wpi.first.wpiutilwpiutil-cppheaders UP-TO-DATE
:downloadedu.wpi.first.wpiutilwpiutil-cpplinuxnativearm UP-TO-DATE
:ntcoreJNIjniHeaders UP-TO-DATE
:compileNtcoreJNISharedLibraryNtcoreJNICpp UP-TO-DATE
:linkNtcoreJNISharedLibrary UP-TO-DATE
:ntcoreJNISharedLibrary UP-TO-DATE
:nativeTestFilesJar UP-TO-DATE
:run
Hello World!
/linux/armhf/
Exception in thread "main" java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
        at java.lang.Runtime.loadLibrary0(Runtime.java:870)
        at java.lang.System.loadLibrary(System.java:1122)
        at edu.wpi.first.networktables.NetworkTablesJNI.<clinit>(NetworkTablesJNI.java:53)
        at edu.wpi.first.ntcore.DevMain.main(DevMain.java:10)
:run FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':run'.
> Process 'command '/usr/lib/jvm/java-8-oracle/bin/java'' finished with non-zero exit value 1

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 3s
9 actionable tasks: 2 executed, 7 up-to-date
```

PeterJohnson, 2017-09-30:
Thanks!  @ThadHouse please take a look

PeterMitrano, 2017-11-06:
update on this? I'm trying to do the same thing.

ThadHouse, 2017-11-06:
This was fixed. Do you have a fresh pull of wpiutil, built and published that, and then attempting to build this?

PeterMitrano, 2017-11-06:
Yep that worked. Thanks! I forgot to publish

PeterJohnson, 2018-05-02:
As ntcore is now part of allwpilib, this is no longer an issue.

closed 2018-05-02 by PeterJohnson