wpilibsuite/ntcore #244: Fix Travis CI building ntcore

JLLeitschuh, 2017-10-04:
As a part of fixing https://github.com/wpilibsuite/shuffleboard/pull/220 I figured out how to harnes shuffleboard into an enviroment where gradle is executing inside of a Ubuntu 16.2 OS.

I think I can take this same basic knowledge and get network tables running under docker as well.

Some things that I would find useful:
 - What are the known `apt` dependencies for building ntcore?
 - Are there other linux distros that there is interest building/testing under (eg. fedora).
 - While I'm at it I may be able to try to get the arm toolchains into a docker container as well so that travis can test with both. (Using the build matrix).

PeterJohnson, 2017-10-04:
RoboRIO ARM toolchain is here: https://launchpad.net/~wpilib/+archive/ubuntu/toolchain-beta
You'll need to add:
```
sudo add-apt-repository ppa:wpilib/toolchain-beta
sudo apt-get update
sudo apt-get install frc-toolchain
```

PeterJohnson, 2017-10-04:
Distro wise, I think we only care about Ubuntu 16.04 right now.  There's really no reason it shouldn't work on other distros, so I'm not seeing an advantage to actually doing CI testing on them.

PeterJohnson, 2017-10-04:
apt dependencies: g++-multilib, lib32stdc++6, python3, python3-pip, clang-format-3.8

Also want to run: `pip3 install --user wpiformat`

JLLeitschuh, 2017-10-04:
> RoboRIO ARM toolchain is here

Do I need to add it regardless of what the compiler target is (Linux desktop vs arm)?

ThadHouse, 2017-10-04:
Technically no, but if we can only maintain a single docker image that'd be better for maintaining

PeterJohnson, 2017-10-04:
By default gradle builds both.  There's flags to disable either arm or native, but why pay the startup time twice?

JLLeitschuh, 2017-10-04:
I'm wondering if it is faster to build each in parallel using the build matrix? If you don't think so I can make it one Docker image. If building each one sequentially is significantly slower then it may make sense to create two travis jobs.

PeterJohnson, 2017-10-04:
Well, it makes sense to make it one Docker image either way right?  We can always decide to build it as separate jobs with the same Docker image?

JLLeitschuh, 2017-10-04:
The docker image, (if not uploaded to the docker hub or similar) will get re-built for every test. I don't know how long it takes to install the arm compiler.

PeterJohnson, 2017-10-04:
ah.  in that case, yes, it may be faster to be two in parallel with the build matrix.  The flags are `-PskipAthena` for native only, `-PonlyAthena` for roborio only.  The roborio only build only needs the frc-toolchain package, the native one needs gcc-multilib, python3, etc.

JLLeitschuh, 2017-10-06:
Command:
```
./gradlew build -PskipAthena --stacktrace --console=plain
```

```
:linkNtcoreSharedLibrary/usr/src/app/build/objs/ntcore/shared/ntcoreCpp/czv21mas3ztwkli314ii733dt/Notifier.o: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
```

Dockerfile:
```
FROM ubuntu:16.04

RUN \
  apt-get update && \
  apt-get install -y software-properties-common curl g++-multilib lib32stdc++6 python3 python3-pip clang-format-3.8 && \
  echo oracle-java8-installer shared/accepted-oracle-license-v1-1 select true | debconf-set-selections && \
  add-apt-repository -y ppa:webupd8team/java && \
  apt-get update && \
  apt-get install -y oracle-java8-installer && \
  rm -rf /var/lib/apt/lists/* && \
  rm -rf /var/cache/oracle-jdk8-installer && \
  pip3 install --user wpiformat

ENV JAVA_HOME /usr/lib/jvm/java-8-oracle

RUN mkdir -p /usr/src/app

WORKDIR /usr/src/app

COPY . /usr/src/app
```

Thoughts??

PeterJohnson, 2018-05-02:
OBE by allwpilib merge.

closed 2018-05-02 by PeterJohnson

JLLeitschuh, 2018-05-02:
OBE?

AustinShalit, 2018-05-02:
OBE = Overcome By Events

AustinShalit, 2018-05-02:
https://github.com/wpilibsuite/ntcore/pull/209#issuecomment-318927155