wpilibsuite/ntcore #265: What to do after publishing in Maven locally?

GillianYue, 2018-01-26:
Hi, 

Our team is trying to hook up NetworkTable in one of the Java programs we built besides the main robot program, and we're having trouble properly getting access to the ntcore library in Eclipse. 

After following the instructions (building using gradlew, publishing to releases/maven/development), we are stuck, not knowing how to make use of what we have at hand. Under the ntcore-java folder, there is another folder titled "4.424242.0.0-20180124161451-dirty", in which three jar files and a bunch of other files exist. How do you establish in Eclipse a local Maven dependency with those files, and what are artifact/group IDs? 

Thank you!

PeterJohnson, 2018-01-26:
Rather than trying to get maven to hook up in eclipse, you're probably just better off downloading the release .jars and adding them to your project.  See https://www.chiefdelphi.com/forums/showpost.php?p=1726695&postcount=6 for download links.

GillianYue, 2018-01-27:
That solved my problem. Thank you so much!

closed 2018-01-27 by GillianYue