wpilibsuite/ntcore #266: segfault in OutlineViewer.jar

virtuald, 2018-02-06:
Linux, openjdk + openjfx, but the segfault is in ntcore somewhere. Using the jar file distributed with the eclipse plugins. Running in server mode.

```
(gdb) bt
#0  0x00007fff8e4c2e73 in llvm::StringMapImpl::LookupBucketFor(llvm::StringRef) () at /lib64/libLLVM-5.0.so
#1  0x00007fff84e3ab1c in nt::Storage::GetOrNew(llvm::Twine const&) () at /tmp/libNetworkTablesJNI822694554168522073.so
#2  0x00007fff84e3bb65 in nt::Storage::ProcessIncomingEntryAssign(std::shared_ptr<nt::Message>, nt::INetworkConnection*) ()
    at /tmp/libNetworkTablesJNI822694554168522073.so
#3  0x00007fff84e3d9a9 in nt::Storage::ProcessIncoming(std::shared_ptr<nt::Message>, nt::INetworkConnection*, std::weak_ptr<nt::INetworkConnection>) () at /tmp/libNetworkTablesJNI822694554168522073.so
#4  0x00007fff84e2fc45 in nt::DispatcherBase::ServerHandshake(nt::NetworkConnection&, std::function<std::shared_ptr<nt::Message> ()>, std::function<void (llvm::ArrayRef<std::shared_ptr<nt::Message> >)>) () at /tmp/libNetworkTablesJNI822694554168522073.so
#5  0x00007fff84e34bea in std::_Function_handler<bool (nt::NetworkConnection&, std::function<std::shared_ptr<nt::Message> ()>, std::function<void (llvm::ArrayRef<std::shared_ptr<nt::Message> >)>), std::_Bind<std::_Mem_fn<bool (nt::DispatcherBase::*)(nt::NetworkConnection&, std::function<std::shared_ptr<nt::Message> ()>, std::function<void (llvm::ArrayRef<std::shared_ptr<nt::Message> >)>)> (nt::DispatcherBase*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)> >::_M_invoke(std::_Any_data const&, nt::NetworkConnection&, std::function<std::shared_ptr<nt::Message> ()>&&, std::function<void (llvm::ArrayRef<std::shared_ptr<nt::Message> >)>&&) ()
    at /tmp/libNetworkTablesJNI822694554168522073.so
#6  0x00007fff84e22881 in std::function<bool (nt::NetworkConnection&, std::function<std::shared_ptr<nt::Message> ()>, std::function<void (llvm::ArrayRef<std::shared_ptr<nt::Message> >)>)>::operator()(nt::NetworkConnection&, std::function<std::shared_ptr<nt::Message> ()>, std::function<void (llvm::ArrayRef<std::shared_ptr<nt::Message> >)>) const () at /tmp/libNetworkTablesJNI822694554168522073.so
#7  0x00007fff84e1fc51 in nt::NetworkConnection::ReadThreadMain() () at /tmp/libNetworkTablesJNI822694554168522073.so
#8  0x00007fff84e22419 in std::thread::_Impl<std::_Bind_simple<std::_Mem_fn<void (nt::NetworkConnection::*)()> (nt::NetworkConnection*)> >::_M_run() () at /tmp/libNetworkTablesJNI822694554168522073.so
```

I'm connecting to the server using RobotPy's cscore, via `python3 -m cscore`

virtuald, 2018-02-06:
Hm. Why did it decide to use /lib64/libLLVM-5.0.so?

PeterJohnson, 2018-02-06:
You’re picking up the wrong LLVM library somewhere.

virtuald, 2018-02-06:
That's weird, shouldn't that be statically linked in? Maybe gdb is just confused.

Unzipping the OutlineViewer.jar shows only libntcore.so files:

```
$ find -name '*.so'
./linux/athena/libntcore.so
./linux/x86-64/libntcore.so
./linux/x86/libntcore.so
```

```
$ ldd linux/x86-64/libntcore.so 
ldd: warning: you do not have execution permission for `linux/x86-64/libntcore.so'
	linux-vdso.so.1 (0x00007fff3b4aa000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f25f3e4a000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f25f3c33000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f25f3a14000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f25f3631000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f25f32dc000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f25f44c9000)
```

PeterJohnson, 2018-02-08:
Something in Java is likely dynamically loading it as part of a graphics library (e.g. Mesa).  We've had one other user report this as well.  The correct fix will be to namespace our variant of llvm's libraries to avoid the name collisions, but I think that's too high risk to do until after the end of the competition season.

PeterJohnson, 2018-04-30:
Will be fixed by wpilibsuite/allwpilib#1006.

PeterJohnson, 2018-05-02:
wpilibsuite/allwpilib#1006 has been merged, so closing.

closed 2018-05-02 by PeterJohnson