wpilibsuite/ntcore #268: Can't build on raspian

redsil, 2018-02-11:
BinaryContainer, ProjectLayout, BuildConfigSpec)
   > Could not resolve all files for configuration ':nativeDeps'.
      > Could not find wpiutil-cpp-linuxnativearm.zip (edu.wpi.first.wpiutil:wpiutil-cpp:3.0.0-20180209143927-2-g4e9e7ec).
        Searched in the following locations:
            http://first.wpi.edu/FRC/roborio/maven/development/edu/wpi/first/wpiutil/wpiutil-cpp/3.0.0-20180209143927-2-g4e9e7ec/wpiutil-cpp-3.0.0-20180209143927-2-g4e9e7ec-linuxnativearm.zip

Appears the cpp binaries don't exist at that location?

PeterJohnson, 2018-02-11:
Building with gradle on other platforms is not currently supported, as we don't distribute binaries for those platforms.  You can use the following repo to build using cmake instead of gradle: https://github.com/ThadHouse/CmakeWpilib

PeterJohnson, 2018-05-02:
Will be fixed by wpilibsuite/allwpilib#1015, which brings back cmake builds.

closed 2018-05-02 by PeterJohnson