wpilibsuite/ntcore #273: Cannot use C interface with gcc

rjbell4, 2018-03-16:
Despite appearing to support a C API, the `ntcore_c.h` file is not usable with a C compiler.  That's primarily because it `#include`s `<cstddef>`, and because it presumes that `struct`s are implicitly `typedef`ed.

ThadHouse, 2018-03-16:
The reason is the c API is primarily for interop between other languages. So it's more to create an API with the C abi, rather then be directly usable from C itself. Is there a specific reason you need to use the header from C?

rjbell4, 2018-03-16:
We are running a C-based program on co-processor for use in FRC.  Recent developments have us wanting to be able to take action based on a state determined by the robot.  So we want the robot to write to a Network Table, and the C-based program it be able to pick up those change.  Therefore, we'd like to be able to read from Network Tables in C.

The changes are seem pretty simple; they appear to just be oversights.  I create PR #274 for the changes I made that allow be to compile the C code with gcc, instead of g++.

rjbell4, 2018-03-16:
Release build issue?  I can't get through a release build with warnings (treated as errors) on my machine _without_ my modification.  Try a slightly less less invasive change, though, to see if that fixes the issue...

ThadHouse, 2018-03-16:
That issue happens on Travis, and we haven't been able to fix it yet because of the old release of gcc Travis has. I'll test it locally and make sure it builds, then we can push it to build on our main servers.

rjbell4, 2018-03-16:
Hmm, OK.  I moved the more C-stuff inside the `__cplusplus` `ifdef`, just in case.  Let me know if you want it taken back out and made universal.

PeterJohnson, 2018-05-02:
Fixed in wpilibsuite/allwpilib#1007.

closed 2018-05-02 by PeterJohnson